dojohnso/OctoPrint-SpoolManager #12: No button to add new spools

hobbieman, 2023-12-09:
I installed via:

~/OctoPrint/venv/bin/pip install https://github.com/dojohnso/OctoPrint-SpoolManager/releases/latest/download/main.zip

I am unable to add any spools as I don't see a button anywhere to do so.

[image]

hobbieman, 2023-12-09:
BTW, I tried this with 3 different web browsers (chrome, safari and edge)... same results

broderickhyman, 2023-12-09:
@hobbieman The spools are added and managed via a Spools tab in the main window
[image]

hobbieman, 2023-12-09:
Doh! Thank you!

closed 2023-12-09 by hobbieman