dojohnso/OctoPrint-SpoolManager #19: Bug - When using Consolidate Temp Control plugin

OnceWasPerfect, 2023-12-13:
Not sure if its with SpoolManager or the Consolidate Temp Control plugin but if the Consolidate Temp Control plugin is set to horizontal the Spool management window gets sized wrong.  If I set it to vertical Spool management goes back to normal.
[image]

dojohnso, 2023-12-13:
we can take a look at some point. likely a little bit of both, but we should be able to find a way to accommodate for that. 

i noticed some... odd choices with the UI on the printjobhistory plugin, similar stuff here. I'm hoping to give both a little refresh at some point.

OnceWasPerfect, 2023-12-13:
Awesome, thanks for taking up the project!

closed 2024-02-04 by github-actions[bot]