dojohnso/OctoPrint-SpoolManager #20: Plugin still not visible

ronzelver, 2023-12-13:
I have installed this fork of the plugin using the built-in Plugin Manager, but after restarting it still doesn't show up in the UI?!?
I have the checked the logfile of the plugin manager and it didn't throw any errors:

`2023-12-13 21:44:40,354 > Building wheels for collected packages: OctoPrint-SpoolManager
2023-12-13 21:44:40,354 > Building wheel for OctoPrint-SpoolManager (setup.py): started
2023-12-13 21:44:40,354 > Building wheel for OctoPrint-SpoolManager (setup.py): finished with status 'done'
2023-12-13 21:44:40,354 > Created wheel for OctoPrint-SpoolManager: filename=OctoPrint_SpoolManager-1.7.3-py3-none-any.whl size=659851 sha256=0e07c0159c36ec92695428db9b8fb29da7a812fd0bd1fdd34cd9334db925d01f
2023-12-13 21:44:40,354 > Stored in directory: /tmp/pip-ephem-wheel-cache-3d2zu70e/wheels/a1/72/5a/898db2801650333f7163cc8783efae030afe08540cec9d9e86
2023-12-13 21:44:40,354 > Successfully built OctoPrint-SpoolManager
2023-12-13 21:44:40,354 > Installing collected packages: OctoPrint-SpoolManager
2023-12-13 21:44:40,354 > Attempting uninstall: OctoPrint-SpoolManager
2023-12-13 21:44:40,355 > Found existing installation: OctoPrint-SpoolManager 1.7.3
2023-12-13 21:44:40,355 > Uninstalling OctoPrint-SpoolManager-1.7.3:
2023-12-13 21:44:40,355 > Successfully uninstalled OctoPrint-SpoolManager-1.7.3
2023-12-13 21:44:40,355 > Successfully installed OctoPrint-SpoolManager-1.7.3
`
OctoPrint 1.9.3  Python 3.9.16

dojohnso, 2023-12-13:
these seems odd..
> Attempting uninstall: OctoPrint-SpoolManager
> Successfully uninstalled OctoPrint-SpoolManager-1.7.3
> Successfully installed OctoPrint-SpoolManager-1.7.3

it would seem for some reason the install process is uninstalling itself... did you install fresh, or did you switch when the update came out?

ronzelver, 2023-12-13:
I did a re-install attempt after unsuccessful attempts with the original plugin. I can try to uninstall it completely using the command line.

dojohnso, 2023-12-14:
> I did a re-install attempt after unsuccessful attempts with the original plugin. I can try to uninstall it completely using the command line. Any suggestions how I can remove an unregistered (but still installed ) plugin from my system (FreeBSD 13.0 Jail)?

I would try something like this. https://community.octoprint.org/t/i-installed-a-broken-plugin-how-do-i-uninstall-it-again/217

ronzelver, 2023-12-14:
I was able to manually delete the existing SpoolManager plugin folders and now the plugin manager log shows:

2023-12-14 23:48:48,944 > Building wheels for collected packages: OctoPrint-SpoolManager
2023-12-14 23:48:48,944 > Building wheel for OctoPrint-SpoolManager (setup.py): started
2023-12-14 23:48:48,944 > Building wheel for OctoPrint-SpoolManager (setup.py): finished with status 'done'
2023-12-14 23:48:48,944 > Created wheel for OctoPrint-SpoolManager: filename=OctoPrint_SpoolManager-1.7.3-py3-none-any.whl size=659851 sha256=9d3870c063562397bb3437a33f87c07f85c2f86b28889f8608a779a45b2a86e4
2023-12-14 23:48:48,944 > Stored in directory: /tmp/pip-ephem-wheel-cache-d22vm198/wheels/2f/e7/2b/f1d84a33d771029f87226169a0fc31b940ef0497e03b83aaac
2023-12-14 23:48:48,944 > Successfully built OctoPrint-SpoolManager
2023-12-14 23:48:48,944 > Installing collected packages: OctoPrint-SpoolManager
2023-12-14 23:48:49,447 > Successfully installed OctoPrint-SpoolManager-1.7.3

So from a build/installation point of view everything seems ok, however the plugin still doesn't show up in the Plugin section?  I have also checked the main octoprint.log and there I see issues with sqlite3:

2023-12-14 23:49:06,393 - octoprint.plugin.core - ERROR - Error loading plugin SpoolManager
Traceback (most recent call last):
  File "/root/Octoprint/lib/python3.9/site-packages/octoprint/plugin/core.py", line 1294, in _import_plugin
    module = _load_module(module_name, spec)
  File "/root/Octoprint/lib/python3.9/site-packages/octoprint/plugin/core.py", line 52, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/root/Octoprint/lib/python3.9/site-packages/octoprint/vendor/imp.py", line 238, in load_module
    return load_package(name, filename)
  File "/root/Octoprint/lib/python3.9/site-packages/octoprint/vendor/imp.py", line 212, in load_package
    return _load(spec)
  File "<frozen importlib._bootstrap>", line 711, in _load
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/root/Octoprint/lib/python3.9/site-packages/octoprint_SpoolManager/__init__.py", line 12, in <module>
    from octoprint_SpoolManager.DatabaseManager import DatabaseManager
  File "/root/Octoprint/lib/python3.9/site-packages/octoprint_SpoolManager/DatabaseManager.py", line 9, in <module>
    import sqlite3
  File "/usr/local/lib/python3.9/sqlite3/__init__.py", line 57, in <module>
    from sqlite3.dbapi2 import *
  File "/usr/local/lib/python3.9/sqlite3/dbapi2.py", line 27, in <module>
    from _sqlite3 import *
ModuleNotFoundError: No module named '_sqlite3'

ronzelver, 2023-12-14:
Problem fixed by installing sqlite3 for python 3.9.18

closed 2023-12-14 by ronzelver