dojohnso/OctoPrint-SpoolManager #29: Merge external db

Hottabych-, 2024-01-08:
Hi @dojohnso ,

could you merge the external db https://github.com/dojohnso/OctoPrint-SpoolManager/pull/17 into the SpoolManager please, or are there any other issues with that?
If interested can help to test this new feature to save some time on your side.

Best regards

dojohnso, 2024-01-23:
hey @Hottabych- , sorry i've been silent here, work has been absolutely nuts lately.

i'll take a look at this today and hopefully get it merged into the dev branch so folks can pull from that channel in Octoprint and test it out to start. make sure you set your release channel on the Settings -> Octoprint -> Software Update section to "Release & In Development"

closed 2024-01-23 by dojohnso

Hottabych-, 2024-01-24:
Thanks a lot @dojohnso, at the moment I'm on a business trip - but the weekend will come ;-)

Hottabych-, 2024-02-18:
Hi, do you see my test protocol?