dojohnso/OctoPrint-SpoolManager #33: [Bug] No Spool Selected

HighTemplar1967, 2024-01-16:
I am getting the below error when printing now. It was giving this error occasionally, but swapping spools would resolve it eventually. I have tried wiping the data, uninstall/reinstall and the issue persists.

[image]

drewoprea, 2024-01-29:
I also get this error every "first time I try to print a uploaded file."
Previously I was just mashing Cancel and trying again until it worked, but then I noticed that it would refuse to work until after the PrintTimeGenius plugin finishes parsing the file (and the flashing star next to my filename becomes solid)

I don't know if PrintTimeGenius parsing the file is a requirement for SpoolManager to work, or if it just has access to the file first, but waiting for it to finish seems to solve this problem on my octoprint.

evilDave, 2024-02-09:
I am seeing the same issue, just started maybe a day ago - just after I updated orcaslicer.

I looked at the code now generated by orca and it contains duplicated metadata. These values:
```
; filament_density: 1.24
; filament_diameter: 1.75
```
are repeated at the top and bottom of the file. I have confirmed that removing these, does indeed fix the issue.

I've logged a bug over there, because technically I guess one should not duplicate the metadata: https://github.com/SoftFever/OrcaSlicer/issues/4020
However it would be nice to be robust against this sort of thing.
I also do not see anything in any of the logs about any error...

chemixt, 2024-02-11:
I love this pluggin, and had been using it for a few years on my Makerbot R2X. But I started getting the same error message about a year ago with 1.7.3 (same with 1.7.4) I am using the GPX plugin to convert gcodes for the MakerBot.  But recently I discovered the issue is gone if I slice (to gcode) with Prusa slicer, or Cura slicers!   But I normally use the MakerBot Desktop slicer. Does anyone know what meta data the spool manager is  looking for in the gcode? I do not see the term density or diameter or M200 gcode in either gcode file. (I am not a programmer)

If it helps attached is a gcode file from Cura, and another from Makerbot slicer for the same item.

[Cura test gcode.txt](https://github.com/dojohnso/OctoPrint-SpoolManager/files/14233893/Cura.test.gcode.txt)

[Makerware test gcode.txt](https://github.com/dojohnso/OctoPrint-SpoolManager/files/14233888/Makerware.test.gcode.txt)

closed 2024-03-24 by github-actions[bot]

evilDave, 2024-07-11:
I realise now that this is not the right place for this at all, the issue has been raised in orcaslicer, and it should be in octoprint to be more robust there to duplicate metadata (I thought I did that but it looks like maybe I didn't). there is nothing that can be done from the plugin which is just hoping to receive the metadata.

I finally got round to writing a post processing script to repair the gcode, that script is here https://github.com/evilDave/patch-slicer-metadata/blob/main/patch-slicer-metadata.py hopefully that might help someone.