dojohnso/OctoPrint-SpoolManager #38: Use absolute temperatures instead of offset temperature overrides

numericOverflow, 2024-03-11:
Is is possible to have SpoolManager submit the absolute temperatures stored on spool profile instead of the offset temp? 

I often want to print something in a new material, but I have to first manually calculate the difference between the original temp set in the GCODE and update the spool's offset.  It would be **MUCH** easier if I could just set a desired temp for the spool and have it submit that temp as an absolute override temp rather than an offset.

If that's not possible for some reason (say if you can only send offsets and not absolute values), could you dynamically calculate the offset on the fly and send the calculated offset?  Just subtract the GCODE's temp(s) from the spool's absolute temp and send that value?  Maybe even have a toggle button in the spool config screen to choose which value is used for that spool (absolute or offset temps).

[image]

numericOverflow, 2024-05-12:
Bump to avoid autoclose.

numericOverflow, 2024-05-30:
Bueller, ...Bueller???

AlexVercammen, 2024-06-27:
I have something similar. In most cases, I setup everything in Cura as PLA for a standard. Therefore, I add a bed and nozzle offsets  if the spool material requires different settings, like PLA-CF. However, in the event, if I do setup Cura to the actual material like PLA-CF, the Gcode temperatures will already be set correctly and I won't need the offset from the plugin to be applied. Could we have the option to SKIP added to OK and CANCEL?