dojohnso/OctoPrint-SpoolManager #47: CSV Backup fails

ccle610, 2024-05-17:
Downloading the CSV File does not start and end up in timeout.

http://<my IP>/plugin/SpoolManager/exportSpools/CSV

M3nsch, 2024-06-23:
I have the same problem.

- Smartphone (Xiaomi Redmi Note 9 Pro)
- octo4a OctoPrint on Android (Octoprint: 1.10.2)
- Spool Manager Version: 1.7.3

jyourstone, 2024-07-19:
Same issue here.

Ruinning OctoPrint version 1.10.2 and SpoolManager version 1.7.3.

ETA-develop, 2024-07-21:
I have the same problem as well.   I think this happened with the octoprint 1.10 update.   I have Octoprint 1.10.2 and SpoolManager 1.7.3.   I use the export to generate automated printed labels for the spools.  I am half way considering rolling back Octoprint but I fear that would cause additional issues in general.

chemixt, 2024-08-21:
I have the same problem and have also entered into newer issue #63

I have the same problem. When I click on the "CSV-File " URL within the plugin, It goes to : "http://octopi.local/plugin/SpoolManager/exportSpools/CSV?instance=internal" in my browser and simply spins saying "Loading".

Spool Manager : 1.7.4
Octopi 0.17.0
Octoprint 1.10.2