dojohnso/OctoPrint-SpoolManager #54: Has this been abandoned?

rtrout, 2024-07-28:
I know it's hard work, but I'm not seeing signs of life. Are the maintainers still here?

rtrout, 2024-09-04:
Looks like it has. Details here:
https://community.octoprint.org/t/is-spool-manager-still-in-development/60025

dojohnso, 2024-09-04:
hey all, yeah turns out i don't have the time to keep up with all the tickets between this and [PrintJobHistory](https://github.com/dojohnso/OctoPrint-PrintJobHistory). if anyone wants to take this over, feel free to do so.

dpenezic, 2024-09-09:
@dojohnso , maybe I am not looking in right place but your git project doesn't have any license? according to github policy, in that case, only viewing and forking on github is permitted.  
Which mean I am using it illegally at a moment ;)

ThePreviousOne, 2024-09-09:
@dpenezic I haven't read the github policy but anywhere else no license file is taken to mean CC0

dpenezic, 2024-09-09:
@ThePreviousOne , please check follow document https://docs.github.com/en/repositories/managing-your-repositorys-settings-and-features/customizing-your-repository/licensing-a-repository , additionally for any other place no license mena no exact rules for using, if CCO in your message is https://creativecommons.org/public-domain/cc0/ , I think that license isn't address to software product at all.
Finally, it isn't too hard to add some license file to the existing git project and everything else will be fine and people may continue development on add-ons without worrying if is it or isn't legal ... same play to the user of plug-in ...