dojohnso/OctoPrint-SpoolManager #57: [Bug] No Spool Selected (former issue #33, but this was never resolved.)

Jartoos, 2024-08-24:
I still have this issue, I was never able to use the spool manager plugin because of it.
[image]

I wil try installing the PrintTimeGenious plugin to see if this resolves the issue.

WildRikku, 2024-09-11:
Could you give more information on when this issue occured? I often have it when I upload a Gcode file and immediately click "load & print". The reason is that Octoprint needs some time (sometimes 10 seconds or more) to process the Gcode internally and as long as it is not done, SpoolManager can't calculate the filament amount.

Jartoos, 2024-09-11:
> Could you give more information on when this issue occured? I often have it when I upload a Gcode file and immediately click "load & print". The reason is that Octoprint needs some time (sometimes [...]

The issue is solved for me.
I found in another thread that you could solve this issue by installing the following plugin first : 
"PrintTimeGenius Plugin"
As soon as I had this plugin installed, the Spoolmanager worked without further issue.

Jartoos, 2024-09-11:
The issue is solved for me.
I found in another thread that you could solve this issue by installing the following plugin first : 
"PrintTimeGenius Plugin"
As soon as I had this plugin installed, the Spoolmanager worked without further issue.

closed 2024-09-11 by Jartoos